sumotoy/TFT_ST7735 #1: 1.1b1 ESP8266 examples not works

reaper7, 2016-02-28:
I try to test 1.1b1 branch on esp8266 but nothing happens on display
(examples: basicSetup, Wifi_scan)
Display connected as in example:
cs=16, dc=5, rst=4, sck=D5, mosi=D7

But when I change library to Adafruit_ST7735, without changing hardware connection
everything works as expected.

btw. I try on nodemcu and wemos boards.

sumotoy, 2016-02-28:
Ok, time to test it really on my ESP. I will fix this in some hours max, must be something stupid I've missed.

reaper7, 2016-02-28:
probably, but easy...weekend still in progress :)

sumotoy, 2016-02-29:
As I expected, was something really stupid. The library works as is, the problem was the max spi speed supported by ST7735, the ILI chip support 80Mhz and more but ST support max 65Mhz, this is why.
I'm not sure if there's any resistors in series in the ST module, chinese like to do such weird stuff like that.
New version online

closed 2016-02-29 by sumotoy

reaper7, 2016-02-29:
now is OK 
:+1: 

P.S. please fix symbol ")" inside fonts arial_x2 because it's looks like "|)"