sumotoy/TFT_ST7735 #6: Space character not rendered when using 'minipixel' font

betterbits, 2016-06-18:
Hi, I have a display based on this chip connected to an Arduino Uno with hardware SPI.
It works well. Very fast.
Not sure if it is the library or what, but when using the 'internal' font, spaces get inserted into text as expected. When using the included 'minipixel' font, spaces are not inserted into the text printed on screen. Text after a space in the source is printed on the screen directly after the text before the space.
I looked at 'internal.c' and 'minipixel.c' and they both include data (0 byte) for character 0x20.
I am still learning C and programming Arduino.
Any help would be appreciated.

sumotoy, 2016-06-18:
Mmm, you are using an old version of the library!
Please use this one:
https://github.com/sumotoy/TFT_ST7735/tree/1.0p1
The font char is special, since there's no pixel I'm using just the 'width' data and ignore completely the glyph. 
Looking inside the old minipixel font (not used in the new library since font file format it's totally changed) it shows a wifth of 6, should render correctly.
Anyway please try the new version, it's faster and more featured, if you still need minipixel I can reconvert for the new library.

betterbits, 2016-06-18:
Hi Max,

Thanks for taking the time to reply.
I installed the new library you linked and have started using it.

I have observed that some of the fonts produce an error when compiling in
Arduino IDE.
"akashi20" is one example.
Arduino IDE reports "externalFonts:39: error: 'akashi20' was not declared
in this scope" when compiling the "externalFonts" example.
I had a look at the font files and the ones that fail to compile do not
have their font names declared in the file, they just reference a generic
'Fonts'
#if !defined(SUMO_FNT_LIB_Font)
#define SUMO_FNT_LIB_Font

I did a find and replace 'Fonts' with the font name and it compiled OK and
displays text in the font.

akashi20, arial20, msShell14, orbitron14 and titlingCaps seem to be the
fonts affected, the rest compile and display OK.

Hope this helps.

Thanks again for your work on the library.

Don

sumotoy, 2016-06-18:
Thanks for point me this. I've used a translator but failed as I can see...
Already fixed. I'll update in some mins...

closed 2016-06-18 by sumotoy