sumotoy/TFT_ST7735 #9: Does not work with platformio

eni23, 2016-08-20:
Hello

First of all, thanks for this library, i really like it.

Unfortunately the library won't compile with platformio. I allready opened an issue there: https://github.com/platformio/platformio/issues/754

The compilation stops with the following error:
```
.pioenvs/nodemcuv2/TFT_ST7735/TFT_ST7735.h:79:32: fatal error: _fonts/defaultFont.c: No such file or directory
#include _ST7735_DEF_FONT_PATH
^
compilation terminated.
```

Do you have some hints what do to that it will compile under Platformio as well? Thanks

ivankravets, 2016-08-22:
Please close this issue because it is related to https://github.com/platformio/platformio/issues/754 and is fixed in the latest development version.

eni23, 2016-08-22:
@ivankravets i will do. thanks for your effort helping users of platformio, im impressed. thats how you get donations ;)

closed 2016-08-22 by eni23

ivankravets, 2016-08-22:
@eni23 Thanks a lot for a donation! 💰 It's important for PlatformIO Project.

Happy Coding with PlatformIO! 😊