sumotoy/TFT_ST7735 #14: Doesn't work with 1.8" TFT SPI on blue PCB

Anton-V-K, 2017-02-03:
I wasn't able to bring my 1.8" TFT display (ST7735) to life with this library (tried Arduino UNO and WeMos D1 R2), so probably it isn't compatible.
It is a Chinese "piece of art" (bought on AliExpress half a year ago), so I'm not surprised it doesn't work according to standards (Adafruit ST7735 library was unable to control it either).
Finally I've managed to control the display with Ucglib (hardware SPI mode on Arduino UNO works fine).

Anton-V-K, 2017-02-04:
After several attempts I've managed to make the library work on WeMos D1 R2 with my 1.8" TFT display (ST7735) via hardware SPI.
I was confused by the examples which reference CS/DC pins in an "absolute" manner with misleading comments. Here is an example from `\_ESP8266\WiFi_scan\WiFi_scan.ino`:
`#define __CS  16  //(D0)
#define __DC  5   //(D1)

/*
 SCLK:D5
 MOSI:D7
*/

TFT_ST7735 tft = TFT_ST7735(__CS, __DC);
`
Probably the defines work fine on some board, but generally they should be written as:
`#define __CS  D0
#define __DC  D1
`

Meins321, 2020-06-30:
Can you create an example and share it here? Would be nice to include it for beginners