sumotoy/TFT_ST7735 #19: Hardware SPI Pins on ESP8266?

arturgoms, 2017-11-07:
It's possible to use the hardware pins to control this lcd?? 
[image]

I mean this up pins, where is GPIO 9 and 10.
And not the sofwarepins in GPIO 14 and 13

Thank You.

RichyT, 2018-10-08:
I think some of those pins are reserved for the flash memory.

> Note that some boards and modules (ESP-12ED, NodeMCU 1.0) also break out pins 9 and 11. These may be used as IO if flash chip works in DIO mode (as opposed to QIO, which is the default one).

http://esp8266.github.io/Arduino/versions/2.0.0/doc/reference.html

I don't know if this helps much. I'd imagine this library isn't equipped to do it.

tbdltee, 2019-10-11:
I have try ESP-12F with Ardafruit GFX and it's working. But you need to connect to the correct pins. It won't work if you connect to the wrong pin. Here is my setup.
#define TFT_CS          0    // GPIO0 --> if you use SPI on GPIO 9, 10. CS always GPIO0
#define TFT_RST         4   // GPIO4
#define TFT_DC          2   // GPIO2
#define TFT_LED       15   // GPIO15
for RST and DC, you can try other pins but it works for me on this 2 pins. I think it's because of pull-up/pull-down stage at boot.
Because CS always GPIO0, you must handle with care. ESP may boot into flash mode. I connect it without pull-down.
You can see that I try to resue all ESP configuration pins. If you connected to normal GPIO pin, it works for sure.
In the sketch, you must call 
  SPI.pins(6, 7, 8, 0);
before init the TFT.
and you must compile and upload in DIO mode.

Then, I switch to TFT_ST7735 library without any changes or tweak but it's not working. To be able to use, I need to investigate more on proper software setup.
I also doubt that can we use this SPI pin? because it's connect to the internal NAND flash and this lib is using high-speed spi.
Anyway, it does not mean "NO". I'm just not try yet.  Need more experiment with this good library.

I'm use this library with arduino and very happy and now try to use it with esp8266. 

that was my experiment so far.
br/tee